chiahao3/ptyrad #4: Wrong hypertune output folder name when varying dz

HanHsuanWu, 2025-05-24:
Hello! Thank you for releasing this amazing tool! When I tested the hypertune feature, it seems like dz is varying as expected from the log.txt but the name for the output folders stays at dz10. I attached some screen shots below for this issue.

[image]

[image]

[image]

HanHsuanWu, 2025-05-25:
@chiahao3. Thanks again for you detailed reply! Regarding the workflow you suggested, would you recommend varying optimizer and batch_size as well during step 2 and 4? For step 3, do you mean a few hundred iterations?

chiahao3, 2025-05-26:
@HanHsuanWu, No problem! The workflow I proposed was just one example and I imagine the community would surely come up with some better / more efficient approaches. 

For new users I think hypertune with BO provides a more principled approach than random search because it utilizes prior information in a statistical way, but for experienced ptycho researchers they usually have a pretty good feeling of what works well (based on their extensive experience), so often times they can manually iterate through the hyperparameters and reach a good solution within maybe 5-10 trials.

Regarding your questions:

1. I prefer to keep the hypertune parameter space as small as possible, like 2-4 parameters at a time. You probably don't need to do hypertune for optimizers and batch size.

   - I don't think optimizers matter that much for the cases I tested. Adam has been performing exceptionally well, AdamW occasionally works better but they're internally very similar anyway. We tested Adam, AdamW, RMSProp, and SGD with different learning rates in the paper and they all seem similar except SGD, probably because we didn't enable momentum so SGD is significantly slower than all the other optimizers. I'll suggest just stick with Adam. Although PtyRAD supports all PyTorch optimizers, so if you're curious you can still try each of them.
    
   - The batch size controls final reconstruction quality, robustness, and computational efficiency. If we ignore the computational efficiency, then small batch size seems to give better reconstruction quality, while larger batch size is believed to be more robust for low dose data (like 1e2 or 1e3 e-/Ang2). My personal preference is to start from batch size = 32 for a 128x128x128x128 EMPAD dataset, and scale it with respect to your total number of scan positions, scan step size, and maybe dose level. If you already have a decent reconstruction and you want to further push the quality a bit, I would suggest try slowly reducing your batch size.

2. Yes that's a typo. I meant for a few hundred (like 200 - 400) iterations. Even for the multislice PSO example that padded to 256x256, 200 iterations is quite sufficient to reach a good resolution. Although to correct most scan distortion I ended up doing 2000 iterations for the image shown in the paper. 

As a side note, in hypertune mode the hierarchy is :

- database (sqlite3 file)
- study (record of the hypertune run. Each run can be continuing an existing study, or be stored as a different study)
- trial (each trial is a reconstruction with a set of hyperparameters) = reconstruction
- iteration (one full pass of all diffraction pattern)
- update step (1 step means 1 mini-batch, which processes B diffraction patterns simultaneously, B is the batch size)

HanHsuanWu, 2025-06-01:
Just saw the new update! Thank you!

closed 2025-06-13 by chiahao3