chiahao3/ptyrad #5: Recenter Z slices in multislice reconstruction?

chiahao3, 2025-05-28:
> @chiahao3 I am not sure if I should open an new issue for this but is there a way in ptyrad to recenter the z slices? 

 _Originally posted by @HanHsuanWu in [#4](https://github.com/chiahao3/ptyrad/issues/4#issuecomment-2917115183)_

chiahao3, 2025-06-13:
@HanHsuanWu Just want to briefly reply on this matter.

The Z slice recentering is on our TODO list for a while, but I plan to add it with a couple new initialization features altogether:

1. Initialization: Vacuum padding along Z for loaded multislice object (presumably do nD padding)
2. Initialization: Probe defocus shifting along Z for loaded probe
3. Initialization: Image shift for object along Z (equivalent to recentering), or add an obj_crop as well to remove unwanted slices
4. Constraint: Re-center the multislice object along Z (and modify the probe defocus accordingly)

These are pretty useful for multislice so I'm hoping to implement them before MM.

Basically the recentering is needed because the ambiguity of object placement and probe defocus, so if the probe is actively changing defocus, the reconstructed object would be moving along the Z dimension inside the object array.

If you want a proper solution, the steps are:

1. Take your reconstructed object and pad it along Z with vacuum layers (note: you need to pad 1 to amplitude and 0 to phase)
2. Use the propagator to "shift" your reconstructed probe along Z according to your vacuum layers. So if you pad 5 nm on top of the object (assuming your object is concentrated near the first slice), you should roll your probe backward for 5 nm so the "initial probe" is now interacting with the padded vacuum layer.
3. Pass your padded object and df-shifted probe into PtyRAD, likely use a `custom` source and directly modify the loaded `params` dict

If you just want a quick workaround, take the reconstructed probe when your object is more or less center along Z, and use that probe for a new reconstruction, but either (1) fix the probe entirely, (2) reduce the probe learning rate, or (3) delay the probe update. This would allow you to stabilize your object reconstruction with a rather steady probe, and once you have a centered object and a reasonable probe, you can do some more quality refinement.

HanHsuanWu, 2025-06-14:
@chiahao3 Thank you for the suggestion. So far, I just adjusted the initial defocus accordingly, and ptyrad was able to recover the probe and centered the object better. To adjust the number of layer and layer thickness, I start from a completely new object with previously reconstructed probes almost like you mentioned. I guess the down side here is that more computation time is wasted.

chiahao3, 2025-08-21:
@HanHsuanWu The latest PtyRAD v0.1.0b11 has added all the necessary feature for users to continue on previous multislice reconstructions! See [release note](https://github.com/chiahao3/ptyrad/releases/tag/v0.1.0b11) for more information!

closed 2025-08-21 by chiahao3