chiahao3/ptyrad #10: could you give an example of misorientation correction for ptychography?

Hansen311, 2025-06-25:
Hi @chiahao3,

  Thank you for sharing this great software. I have tried few examples you given, the results are excellent. I got a problem when tring the misorientation correction by grad descent. For my understanding, I should open the "loss_simlar" as below:

loss_params : {
    'loss_single': {'state': true, 'weight': 1.0, 'dp_pow': 0.5}, # NRMSE error between single simulated and experimental diffraction pattern. 'dp_pow' is commonly set at 0.5 so NRMSE(DP_sim^0.5 - DP_exp^0.5) is equivalent to the Gaussian noise model for typical dataset (dose-sufficient) under the maximum-likelihood formalism
    'loss_poissn': {'state': false, 'weight': 1.0, 'dp_pow': 1.0, 'eps': 1.0e-6}, # negative log likelihood between simulated and experimental diffraction pattern. 'dp_pow' is commonly set at 1 so - (DP_sim * (DP_exp) - DP_exp) is equivalent to the Poisson noise model for low dose dataset under maximum likelihood formalism. See Odstrˇcil2018 https://doi.org/10.1364/OE.26.003108 for more details
    'loss_pacbed': {'state': false, 'weight': 0.5, 'dp_pow': 0.2}, # NRMSE error between simulated and experimental position-averaved CBED (PACBED). Similar to 'loss_single', except that it's comparing PACBED with PACBED and mostly focusing on the diffuse background when 'dp_pow' is set at 0.2
    'loss_sparse': {'state': true, 'weight': 0.1, 'ln_order': 1}, # L_n norm regularization calculated for object phase. 'ln_order' means the L_n norm (|a_i^n|^(1/n)) used to regularize object phase ('objp'). Usually 'ln_order' is set at 1 for L1 norm (|a|), this promotes 0 in the objp and enhance the sparsity (i.e. discrete atoms). 'ln_order' = 2 would be equivalent to L2 norm that promotes near-0 values
    'loss_simlar': {'state': true, 'weight': 0.1, 'obj_type': 'both', 'scale_factor': [1,0.5,0.5], 'blur_std': 1} # std across omode dimension for obj. This promotes similarity between object modes. 'obj_type' can be either 'amplitude', 'phase', or 'both'. 'scale_factor' as (zoom_z, zoom_y, zoom_x) is used to scale the object before calculating the std, setting 'scale_factor' to [1,0.5,0.5]  is equivalent to downsampling the obj 2x along y and x directions before calculating the std, which should encourage the obj modes to keep lateral atom shifts. Similarly, 'blur_std' applies a 2D (lateral) Gaussian blur kernel with specified std to blur the obj before calculating std along omode dimension
}
and 
model_params : {
    'obj_preblur_std'     : null, # type: null or float, unit: px (real space). This applies Gaussian blur to the object before simulating diffraction patterns. Since the gradient would flow to the original "object" before blurring, it's essentially deconvolving the object with a Gaussian kernel of specified std. This sort of deconvolution can generate sharp features, but the usage is not easily justifiable so treat it carefully as a visualization exploration
    'detector_blur_std'   : 1, # type: null or float, unit: px (k-space). This applies Gaussian blur to the forward model simulated diffraction patterns to emulate the PSF of high-energy electrons on detector for experimental data. Typical value is 0-1 px (std) based on the acceleration voltage 
    'optimizer_params'    : {'name': 'Adam', 'configs': null, 'load_state': null}, # Support all PyTorch optimizer except LBFGS because LBFGS can't set separate learning rates for different tensors. The suggested optimizer is 'Adam' with default configs (null). You can load the previous optimizer state by passing the path of `model.pt` to `load_state`, this way you can continue previous reconstruciton smoothly without abrupt gradients. (Because lots of the optimizers are adaptive and have history-dependent learning rate manipulation, so loading the optimizer state is necessary if you want to continue the previous optimization trajectory)
    'update_params':{
        'obja'            : {'start_iter': 1, 'lr': 5.0e-4}, # object amplitude
        'objp'            : {'start_iter': 1, 'lr': 5.0e-4}, # object phase
        'obj_tilts'       : {'start_iter': 20, 'lr': 0}, # object tilts
        'slice_thickness' : {'start_iter': 10, 'lr': 0}, # object dz slice thickness 
        'probe'           : {'start_iter': 1, 'lr': 1.0e-4},  # probe
        'probe_pos_shifts': {'start_iter': 1, 'lr': 5.0e-4}, # sub-px probe positions
    }
}

however, I got error as below: "**RuntimeError: torch.linalg.eig: input tensor should not contain infs or NaNs.**"

So, I wonder if you could offer an example to do the misorientation correction by grad descent? Thank you in advance.

chiahao3, 2025-06-25:
Hi @Hansen311 , thanks for trying out PtyRAD!

`loss_simlar` is designed for mixed-state object as described in the comments so **you should keep `loss_simlar` False if you're not reconstructing multiple object modes.**
```yaml
'loss_simlar': {'state': false, 'weight': 0.1, 'obj_type': 'both', 'scale_factor': [1,0.5,0.5], 'blur_std': 1} 
```

The observed RuntimeError is likely due to `loss_simlar` is trying to take standard deviation across the object modes dimension, but you can't really take standard deviation if you only have 1 object mode so it's throwing errors.

---

For your intended misorientation correction, PtyRAD supports both (1) fixed value correction and (2) AD-based iterative correction.
The (1) fixed value correction is controlled by specifying the initial tilts:
```yaml
'tilt_params'            : {'tilt_type':'all', 'init_tilts':[[0,0]]},
```
You can specify your desired global tilt (y, x in mrad) to 'init_tilts'. This value will be applied across all probe positions.

The (2) AD-based iterative correction will try to optimize the tilts with gradient descent, and is controlled by `update_params`. However, you'll need to specify both the 'start_iter' and 'lr', where 'start_iter' must be smaller than NITER, and **'lr' must be non-zero.**. The initial tilts are also specified by 'tilt_params'.

 ```yaml
'update_params':{
'obja' : {'start_iter': 1, 'lr': 5.0e-4}, # object amplitude
'objp' : {'start_iter': 1, 'lr': 5.0e-4}, # object phase
'obj_tilts' : {'start_iter': 20, 'lr': 0}, # object tilts
'slice_thickness' : {'start_iter': 10, 'lr': 0}, # object dz slice thickness
'probe' : {'start_iter': 1, 'lr': 1.0e-4}, # probe
'probe_pos_shifts': {'start_iter': 1, 'lr': 5.0e-4}, # sub-px probe positions
}
}
```

---
The AD-based correction is designed for position dependant tilts, so **generally I'll suggest the (1) fixed value correction with global tilts for efficiency**. Additionally, you can often obtain a decent initial guess of the tilt by observing the lateral shifts between different object slices.

My typical workflow is:
1. Do a normal reconstruction
2. Estimate the global object tilt by tracking the lateral shifts between slices
3. Do another reconstruction with the estimated tilt in 'tilt_params'

If the estimated tilt is accurate, the atoms should remain stationary while going along the depth dimension (i.e., straight columns). Usually I repeat step 2-3 twice. See our [Box folder](https://cornell.app.box.com/s/n5balzf88jixescp9l15ojx7di4xn1uo) for a bit more information regarding the sign convention in "20240516_Orientation and sign convention of PtyRAD.pptx".

We're still working on compiling more demo notebooks and documentation for these advanced usages, but hopefully these information and the comments of 'tilt_params' can get you moving again!

Hansen311, 2025-06-26:
Hi @chiahao3 ,

  Thank you very much for your quick response. 

Best wishes，

closed 2025-08-10 by chiahao3