chiahao3/ptyrad #14: Probe constraint

material-JH, 2025-08-10:
Hi @chiahao3,

First of all, thank you for the wonderful software. It seems to work well in most cases. However, I've noticed an issue during reconstruction in some specific case (maybe, In Focus?). One probe is on the center position but the other appear at the edge. So, object also reconstructed not only the center, but also at the edge. So, I want to constraint the probe like below, 

modified from "apply_probe_mask_k"
            probe = model.get_complex_probe_view()
            Npix = probe.size(-1)
            powers = probe.abs().pow(2).sum((-2,-1)) / probe.abs().pow(2).sum()
            powers_cumsum = powers.cumsum(0)
            pmode_index = (powers_cumsum > power_thresh).nonzero()[0].item() # This gives the pmode index that the cumulative power along mode dimension is greater than the power_thresh and should have mask extend to this index
            mask = torch.ones_like(probe, dtype=torch.float32, device=model.device)
            mask_value = make_sigmoid_mask(Npix, relative_radius, relative_width).to(model.device)
            mask[:pmode_index+1] = mask_value
            probe_r = mask * probe
            probe_int = model.get_complex_probe_view().abs().pow(2)
            # Re-sort the probe modes, note that the masked strong modes might be swapping order with unmasked weak modes
            model.opt_probe.data = torch.view_as_real(sort_by_mode_int(probe_r))

chiahao3, 2025-08-10:
Hi @material-JH ,

Thanks for the kind words and trying out PtyRAD!

Would you mind share some information (i.e., the log file, params file, summary figures of the problematic reconstruction), so we can better identify the issue and propose possible solution? It's not clear what kind of artifact you're experiencing given the description.

From the proposed code I assume you want to apply the sigmoid mask in real space for the probe, probably due to some intensity got reconstructed at the corner of real space probe. This corner intensity has been known and observed across all major ptychography packages. Fundamentally, I suspect it's an ambiguity in the phase of the probe wavefunction, and I don't think there's a known solution to this unless we can stablize the probe phase. A working progress is to parameterize the probe directly with aberration coefficients. Another potential solution might be to fit and modify the k-space probe phase directly with aberration coefficients, although the fitting procedure itself is a hard inverse problem.

We did tried constraining the probe in real space back in last year but it didn't really solve this specific issue. Altough you can surely experiment it a bit more and see if it helps in your case.
[CHANGELOG](https://github.com/chiahao3/ptyrad/blob/main/CHANGELOG.md)
> ## [v0.1.0-alpha3.7] - 2024-05-09
> ### Added
> - Add `get_date` and `prefix` to `make_output_folder`
> ### Changed
> - Add new fields like `tilt_obj`, `dk`, and `dx` to `make_save_dict`
> - Fix the condition statements for `self.tilt_obj` in `models` so it behaves correctly when the initialized `obj_tilts` is [0,0] so no need for tilt propagator
> - Fix `kz_filter` to use `torch.real()` for the Fourier filtered obj to respect possible negative values
> - Delete the just-added fftshift/ifftshift in `forward` as they don't seem to matter and the forward pass is 50% slower with additional fftshifts
> ### Removed
> - Remove `fix_probe_com` constraints because constantly shifting global probe would cause instability of the obj. Once the obj is fairly reconstructed, shifting the global probe would require shifting the already reconstructed object (or equivalently the entire (N,2) `probe_pos_shifts`) 
> - Remove `probe_mask_r` constraints because it's just not physical and I can't justify myself clipping so much real-space probe tails
> - Remove `obj_chgflip` constraints because the performance is rather poor
> - Remove `obj_same` constraints because the usecase is rather limited and could probably be replaced with `kz_filter`. Optimizing the broadcasted averaged slice wouldn't be computationally faster. Besides, the "rough" result cann't be used to estimate obj tilts so is less usefull than the obj regularized by `kz_filter`

material-JH, 2025-08-11:
[image] this is one of them. 5th probe is on the edge. This is ongoing reconstruction, I didn't use modified probe constraint.

closed 2025-09-01 by material-JH