chiahao3/ptyrad #21: strange opt pmodes

luisc7794, 2025-08-29:
Hello,

I am having an issue with my data where after about iteration 60, the probe disappears. Does this mean my reconstruction failed and if so what can I do to solve this issue?

Thank you an advance!

[image]

chiahao3, 2025-08-31:
Hi @luisc7794 , thanks for trying out PtyRAD!

Yes, I'd say if your probe disappered then it's an unrealistic reconstruction. The tricky part of ptychography is you might still be able to get a decent object even your probe looks like a hot mess, so that's the core reason why PtyRAD provides all these diagonsis figures to draw users' attention and hoping to promote accountability in the community.

I should've created an issue template for these type of questions, but in order to help you in a constructive way I'd need the following information:

1. Your PtyRAD param .yml file so I can check the values you used
2. Your .txt log output file (They're usually automatically saved to your output folder and named like "20250829_184343_ptyrad_log.txt"). This helps me track how PtyRAD is running step by step, and contains necessary hardware and software information.
3. A zip file of the output figures (.png files starting from "summary_XXX.png", ideally a series of them like iter10, 20, 30, 40, 50...)
4. (Optional) You can email me with a link to your data so I may be able to run some local test and better correct your reconstruction parameters

Since I don't have any information so far, my best guess would be the input params might be very off. Like convergence angles, kV, defocus, scan step size, or even the data orientation. We have a Youtube channel [https://www.youtube.com/@ptyrad_official](https://www.youtube.com/@ptyrad_official) that contains many useful information, and hopefully you've at least tried PtyRAD's demo datasets (the tBL-WSe2 and PSO) to make sure everything is working as expected.

luisc7794, 2025-09-02:
Hi @chiahao3 ,

Thank you for your help and insightful comments. Here is the output folder and param file. I will try remeasuring the convergence angle later today. The kV and defocus should be correct and I also double checked the scan step. There is a scan rotation in our microscope which I calculated using py4DSTEM to be 10 deg and I did use that in the param file. I have tried re-ruring by changing some constrains but have not had success.

The youtube channel was very helpful and I did start withe the PSO demo.

https://drive.google.com/drive/folders/1tm9dZkpoMf3-2dRP3uDo8Q9G7TzYq88R?usp=drive_link

chiahao3, 2025-09-02:
Hi @luisc7794 ,

No problem, here are my suggestions:

**Experiment params**
1. 30 mrad might be a bit too large given your aberration correction state. We can clearly see some fringes near the edge of your aperture coming from high order aberrations. Although theoretically ptychography can correct aberrations, we have detailed discusion in our [paper](https://www.science.org/doi/full/10.1126/science.adl2029) that it's still easier to clip those high angles with your probe-forming aperture because you'll need higher k-space sampling to resolve those fine fringes. It might be easier to go with 15 mrad here. As long as you have the same kMax (collection angle) and reasonable SNR at dark field region, your lateral spatial resolution would remain unchanged.
2. The BF disk is a bit elliptical distorted and we can see the aperture edges being a bit serrated. The elliptical distortion is probably coming from some astigmatism, while the serrated edges is a bit more concerning. You might want to check your optics by acquiring diffraction patterns at vacuum (to check the ellipticity and serrated edges) and presumably on thin amorphous region to check your Ronchigram for aberrations. The serrated edges could be coming from dirty aperture, incorrect gun lens (or monochromator focus) setting such that you don't have enough source magnification at C1 aperture, or other weird optics.
3. I'd recommend to go with some slight overfocus like 10-20nm depending on your convergence angle and sample thickness. This helps with orientation determination and gives much higher probe overlap ratio for stability.
4. There's a slight circular shadow in the bottom of the diffraction pattern and it's probably an annular detector or diaphragm. Ideally you want to remove it if possible, you you'll have to reduce your collection angle. The quick way without retaking the data is to crop it via `'meas_crop'`.

[image]

[image]

**Reconstruction params**
1. `obj_Nlayer` you may go with fewer slices for faster debugging. Typically I go with 10 slices of 10 Ang and slowly increase the layers by `obj_z_pad` to search with optimal thickness. Some people would even do single slice ptychography (obj_Nalyer: 1) just to get a even quicker response. For low-density material with light elements, single slice ptychography can go a long way till 10-20 nm without much issue.
2. `meas_flipT` would need to be correct to give sensible results and as mentioned in my [youtube video](https://youtu.be/ZQQ7Y8uzIS0?si=OW5Jg7IDRKluF2-z&t=164) that the result from incorrect orientation can be quite subtle, so you definitely want to check whether its correct. Worst case scenario is just tesing 8 commutations so really not bad if you go with thinner object and smaller FOV.
3. You can set `meas_crop` to [[0,64], [0,64], null,null] to crop the real space to 64x64 scan positions to reconstruct a smaller FOV for even faster debugging. This would not modify your original file on disk, and you don't need to change any other setting (i.e., keep your `pos_N_scan_slow`, `pos_N_scan_fast`, `pos_N_scans`, `meas_reshape` as is because they'll be automatically updated during `meas_crop`)
4. You probably don't need `meas_pad` because you already have enough kMax. Just set it as `'meas_pad' : {'mode': null, 'padding_type': 'power', 'target_Npix': 256,'value': 0}`
5. I'm not sure how robust the scan rotation estimation routine of py4DSTEM is given factors like SNR, defocus, and most importantly diffraction pattern orientation. Assuming everything is correct, the retrieved py4DSTEM rotation angle has an opposite sign convention then in PtyRAD. If you measured +10 deg in py4DSTEM, you will likely need to do -10 deg in PtyRAD, which is `'pos_scan_affine' : [1,0,-10,0]`
6. This doesn't affect your reconstruction, but the correct way to specify that you don't know the key to your hdf5 file is `'meas_params'  : {'path': 'C:/Users/ladmin/Documents/20250812 NV Irr Diamond/Ptycho_1/Reconstruction/data/SPIM_reshaped_bin2.hdf5', 'key': null}` in your yml. PtyRAD automatically parse the yml entry into valid Python dict so 'null' would be converted to 'None'.
7. I'm not sure which detector you're using, but for EMPAD you don't really need much `'detector_blur_std'` for 100 kV electrons. For other detectors you might want to check their kV-dependent PSF or MTF to determine your `'detector_blur_std'`. Generally 0-1 should be safe for most detectors.

The rest of the reconstruction params all seem fine and I expect these standard params should be fairly robust across a wide range of datasets.

My overall suggestion is to first reduce the model complexity and size by (1) use fewer Nlayer, (2) crop your 4D dataset in real space to get fewer scan positions, (3) turn of `meas_pad`, and just focus on testing the 8 possible permutation of `meas_flipT`.

luisc7794, 2025-09-05:
Hi @chiahao3,

Thank you so much for your help and suggestions. I have spent the last couple days troubleshooting this dataset. 

[image]

I implemented most of your suggestions and see a large improvement in my reconstruction. I am attaching a reconstruction where I increased the slice number and used a  meas_flipT of 100.

[image]

Do you think you would call this reconstruction a success?

I plan to get more data next week where I will implement your experimental suggestions and see if I can improve this even more.

chiahao3, 2025-09-05:
Hi @luisc7794 ,

Thanks for sharing, this is phenomenal and I'm super happy to see that you managed to improve the reconstruction quality significantly! I think you're on the right track and I do think if you can acquire better data it'll be even better. A couple notes for your current reconstruction:

1. The probe is at least updating and not dissipating, which is good!
2. The zsum object phase looks much better, although I suspect there might be some crystal tilt because the atoms seem elliptical. You may confirm whether there's any tilt by summing all your diffraction patterns along scan directions in py4DGUI.
3. The model diffraction pattern looks like an okay match, but you can still see some unphysical halo around the BF disk, which might be coming from the difficulty to model the probe given your complicated aberrations near the aperture edges.

I think if you can improve the aberration correction, use a smaller convergence angle to crop out those aberrated part, and acquire at a slightly larger overfocus (like 20nm) at a scan step size of 0.4 Ang, you should be able to get very decent reconstruciton at your current collection. If anything, consider putting a bit more dose to increase the SNR.

I'll close this issue for now but feel free to re-open it or create new ones!

closed 2025-09-05 by chiahao3