chiahao3/ptyrad #22: Potential Memory leak?

HanHsuanWu, 2025-08-30:
Hello @chiahao3,

There seems to be a memory issue during hypertuning with the new release. "CUDA out of memory" comes up during the loading process after the first trail finishes. I also tried lowering the group size but the issue persists. I think this was not a problem before. The log is attached below.

Cheers,

[ptyrad_output_0_3307283174.log](https://github.com/user-attachments/files/22056079/ptyrad_output_0_3307283174.log)

chiahao3, 2025-08-31:
Hi @HanHsuanWu ,

Thanks for providing the log, it's extremely helpful. 

If possible I'd like to see your job script (i.e., how you initiate the hypertune tasks) and know a bit more about your system like if it's managed via Slurm, or it's a local machine, and is it shared by others, etc. 

I think you might be accidentally assigning all hypertune processes on the same GPU, so each process might be creating a copy of your 4D-dataset (~ 8GB). The reason being:

1. Line 31 we can see that you have 4 GPUs on the same machine, so I suspect you're trying to run hypertune with multiple GPUs. 

Usually when we do hypertune on multiple GPUs via a HPC cluster, we request only 1 GPU per Slurm script, and just submit the same script N times so we have N workers on N different nodes, each node we can acceess only 1 GPU so it's safe to keep the same gpuid  for all of the jobs. See [demo scripts](https://github.com/chiahao3/ptyrad/tree/main/demo/scripts) (`LoopSubmit.sh` and `slurm_run_ptyrad.sub`) for more details.

2. Line 1448 Trial 0 just finished, but then Line 1490 you hit OOM error immediately at **Trial 4**, this suggests that there were 3 more workers writing to the hypertune SQLite3 database simultaneously.

3. Same line in the error message PyTorch saw 3 more processes that are using only 400MB ish, although I'm not sure if they're Pytorch processes. 

In sum, I guess you might be submitting this command `ptyrad run --params_path "${PARAMS_PATH}" --gpuid 0 --jobid "${JOBID:-0}"` 4 times without changing the gpuid, so all the PtyRAD process are running on the same GPU. 8GB data * 4 ~ 32 GPU RAM, which seems to match the 36GB number reported by the error message. 

The above command would work beautifully in a Slurm script when we request 1 GPU per job and we submit the same script multiple times, all pointing to the same SQLite3 database. But if your machine contains more than 1 GPU, what you want to do is:

```bash
ptyrad run --params_path <PARAMS_PATH> --gpuid 0 --jobid "0"
ptyrad run --params_path <PARAMS_PATH> --gpuid 1 --jobid "1"
ptyrad run --params_path <PARAMS_PATH> --gpuid 2 --jobid "2"
ptyrad run --params_path <PARAMS_PATH> --gpuid 3 --jobid "3"
```
First you must specify different GPU id, so each PtyRAD process would do the computation on different GPUs. 
Then it's suggested to provide a distinct jobid, so the standard ptyrad log output would have unique identifiers and won't have multiple workers writing to the same file.

Let me know if it works!

HanHsuanWu, 2025-09-03:
Thank you so much again for taking the time to answer the questions!
I intended to do 1 gpu per trial on our workstation with 4 GPUs. I think I have been submitting jobs each with a different gpuid and jobid (your suggested format).
For some reason the OOM error hasn't been showing up, I will update this when it comes up again.
Thanks again!

chiahao3, 2025-09-04:
Hello @HanHsuanWu ,

No problem, these are valid questions and it reminds me to make a better hypertune tutorial based on possible hardwares.

You can add some printing/echo to your bash script then you can keep a record of which gpuid has been supplied. In principle if you set it up correctly, each PtyRAD process should output its own log file in the hypertune folder. Also note that if it's a shared workstation (i.e., other group member can access it) then it's much easier to have multiple process accessing the same GPU.

The following command is essentially starting 4 independent PtyRAD instances, each of them are assigned with an unique GPU. Because you want them to all run hypertune altogether, so the <PARAMS_PATH> would all be the same hypertune params file like `tBL_WSe2_hypertune.yml`. Since it's the same file, the optimization params, the `storage_path`, and the output directory would all be the same. This is how PtyRAD and Optuna managed to run hypertune on multiple GPUs without actually transferring data between processes, because each GPU (i.e., worker) would report its trial result to the database, and all workers would access the same database before starting a new trial, so they're all aware of the current overall progress while making guesses. I think it's a very clever design.

```
ptyrad run --params_path <PARAMS_PATH> --gpuid 0 --jobid "0"
ptyrad run --params_path <PARAMS_PATH> --gpuid 1 --jobid "1"
ptyrad run --params_path <PARAMS_PATH> --gpuid 2 --jobid "2"
ptyrad run --params_path <PARAMS_PATH> --gpuid 3 --jobid "3"
```

To check whether each GPU is assigned with an unique PtyRAD or python process, you can type in `nvidia-smi` in your command line to check the GPU status, process name (something like `<PTYRAD_ENV_NAME>/bin/python`), and corresponding PID after you initiated each `ptyrad run` command. 

I'd close this for now but if you find issues again feel free to re-open it.

closed 2025-09-04 by chiahao3