chiahao3/ptyrad #27: Grid artifact cause?

rahimr2, 2025-09-17:
Hi,

I was wondering if you were familiar with this artifact/failure in ptyrad:

[image]

Imaged above: Sum of slices of reconstruction, 15 slices of 20 A each.

Specifically the grid-like periodic texture in the image.

Looking at the slices, the first slice in the stack seems to be in the right direction:

[image]

but then the rest of the slices in the stack become background fluctuation with increasingly prominent grid texture:

[image]

I was wondering if you happen to know what causes this? I am seeing this in a couple of my reconstructions, and sometime it goes away but sometimes it doesn't. So far I have tried adjusting slices, slice thickness, batch size, and probe learning rate.

My current hypothesis is that it arises from a probe that is too constrained. (I lowered its lr to avoid probe object mixing. Perhaps there is a better way to avoid it, though, like hyperparameter tuning to find better probe parameters?).

Do you have thoughts on this? I am happy to provide more information on the parameters/dataset if that helps. I started with the PSO .yml file and made the appropriate adjustments to it.

chiahao3, 2025-09-17:
Hi @rahimr2 ,

I've seen these grid artifacts in both PtychoShelves and PtyRAD but honestly not very persistent. Similar to what you've described that some of the dataset / reconstruction params seem to produce more of this, but overall I think this is rather rare and I haven't really seen this in PtyRAD for a while so I wonder how it's been triggered.

I'll suggest the following:

1. Do a full check of your reconstruction params, especially meas_flipT, probe df, convergence angle, dx, scan step size, and use a normal learning rate with 1 Ang slice. Sometimes you can get confusing results with incorrect flip / dx so it's always worth checking it. Look closely into the log output and see if you can spot anything.
2. Zip the output folder with log and param file and send it to me so I can examine the reconstuction trajectory with the summary figures. Often times the object would look terrible when the probe is not learning at all, and it generally happens when we put too much constraint, too slow of a learning rate, or the initial param is just too off.
3. If you don't mind I can also try some quick reconstruction if you zip and send your data along with param files.

chiahao3, 2025-09-28:
Hi @rahimr2 ,

Though we already chatted on Slack, I'll still put some results here for archival purpose and potentially benefit other users or future us.

## Reconstructions with artifacts
So you provided 2 reconstructions:
- `full_N16384_dp256_flipT010_random16_p4_1obj_20slice_dz10_plr5e-6_oalr5e-4_oplr5e-4_slr1e-4_tlr1e-5_orblur1.0_ozblur1.0_mamp0.03_4.0_oathr0.98_oposc_pmk0.44_sng1.0_psn1.0_spr0.1`
- `full_N16384_dp256_flipT010_random256_p4_1obj_20slice_dz10_plr5e-6_oalr5e-4_oplr5e-4_slr1e-4_tlr1e-5_orblur1.0_ozblur1.0_mamp0.03_4.0_oathr0.98_oposc_pmk0.44_sng1.0_psn1.0_spr0.1`

[image]

Note: Top row is the top slices, while bottom row is the bottom slices.

There're surely some gridding artifacts but their intensities are also noticeably weaker than the top slices when we display them at the same scale. I think this is consistent with my impression that we often get similar noisy / griddy artifacts at vacuum region.

## My own reconstructions - Full FOV
Since you kindly provided the 4D-STEM data I'm able to diagnose deeper (and I recommend other users to do the same in the future if you have run into issues).
When I skimmed through your params files, I've modified quite some places including:

1. Adjust the initial probe defocus from -100 Ang overfocus to -50 Ang overfocus as I noticed the object seems to be concentrated near the top slices, also the Ronchigram magnification seems too large for a 10nm overfocus.
2. Turn off `meas_pad` as it generally is not needed, especially for debugging purpose
3. Change the probe learning rate from 5e-6 back to 1e-4 as it's quite important to allow the probe to handle experimental mismatches, I'm suspecting this is the main culprit
4. Turn off `loss_poissn` because it's likely not needed given we already have `loss_single` enabled. Roughly speaking the `loss_single` corresponds to the Gaussian / Amplitude noise model, while `loss_poissn` corresponds to the Poisson noise model.
5. Turn off `probe_mask_k` constraint. This constraint by itself doesn't really improve the reconstruction quality and you'll have to set the cutoff radius rather accurately. I implemented this fairly early during PtyRAD development and I think it'll need to be paired with some mixed object formalism otherwise we can't reproduce the diffuse background properly. Typically those diffraction pattern features are absorbed into the probe (like the Kikuchi bands) but if we apply a tight `probe_mask_k`, then the diffuse background would need to be generated via object and it'll often lead to high frequency noise in the object which is undesirable.
6. Enable `detector_blur_std` of 0.5 px as it seems to help EMPAD reconstructions especially in high kV. There's a comparison in issue #9 for the demo PSO dataset. For 300kV it might be better with 1 px but I haven't tried it yet.

After the modification, I'm able to get this:

[image]

Note: Left panel is the orthogonal views, while right panel is the bottom slice with the same display range.

We can see that there're still some gridding artifact but the intensities are much weaker now. I then noticed that the probe wasn't converging that well and is likely due to (1) The top crystalline part was severly tilted by ~ 30mrad, (2) The middle region is amorphous, and (3) the bottom crystalline part was probably off zone for a few degrees. So I did some `meas_crop` and focus only on the top part to see if we can do a bit better.

[image]

[image]

## My own reconstructions - Small FOV
With a reduced FOV, I also corrected the tilt and modified the defocus to center the object around the middle slice. Although we can still see some probe artifacts, the intensities for those probe modes are significantly lower than before. Also the object quality is significantly improved with a better match between model and exp diffraction patterns.

[image]

[image]

[image]

Note: Top panel is the top slice, middle panel is the orthogonal views, while bottom panel is the bottom slice with the same display range.

Overall speaking, with the reduced FOV I don't really see the gridding artifact anymore, there're surely some spurious intensities in the vacuum region as expected due to the limited depth resolution. In addition, the satellite darker column next to the bright column is probably an artifact from the severly tilted grain as I tried correcting ~ 30mrad of tilt.

## Conclusion
Although I didn't really pin point what was the culprit, I suspect it's the probe learning rate being too small, and the fact that probe isn't converging that well when the FOV contains only amorphous and severely tilted region. In addition, by adjusting initial defocus we can minimize the range of consecutive vacuum region which might play some role as well. I'll close this issue for now but feel free to re-open it or reply if you find out what was the root cause, or if you run into this gridding artifact in other datasets.

[image]

Note: Left: initial probe defocus -50 Ang. Right: initial probe defocus 50 Ang.

closed 2025-09-28 by chiahao3

rahimr2, 2025-09-29:
Super thorough, informative, and helpful response. Thank you so much!