chiahao3/ptyrad #28: Initialization issues - large datasets

emanuelaliberti, 2025-09-22:
Dear Chia-Hao,
I have been testing PtyRAD on large 4D-STEM datasets (>150 Gb) and I am running into issues with the initialization step that seems to require impractically large host memory. For instance, during the Initializer(init_params).init_all(), the function _load_meas casts the entire dataset to float32. The subsequent steps (meas_flitT and meas.mean(0)) also require extremely large host RAM.
Do you have any suggestions? 
Best regards,
Emanuela

emanuelaliberti, 2025-09-22:
Hi @chiahao3,
Thank you so much for getting back to me.
I am sharing the the .yml and log file from a successful run test on a much smaller dataset. I would like to do the same, but for a much larger file approx. 150 GB.
If you have any suggestions please let me know. As you can see I do have access to multiple GPUs (2 or also 4 eventually). Can we avoid pre-processing on the CPU? I managed to compress using .h5 conversion the original data to 14 GB but even this size requires large memory during the pre-processing step because of the float32 cast.
Thanks for your help in advance!

[XuGrat.yml](https://github.com/user-attachments/files/22469024/XuGrat.yml)
[20250910_142334_ptyrad_log.txt](https://github.com/user-attachments/files/22469023/20250910_142334_ptyrad_log.txt)

chiahao3, 2025-09-22:
Hi @emanuelaliberti ,

No problem! Got the files and I'll look closer in this soon. Can you also provide the cell outputs when you try the 150GB dataset? I'd like to see at which step it runs into OOM on your machine.

The float32 cast is required for automatic differentiation, and will likely occur during the preprocessing step for the normalization step but we can look for alternative.

emanuelaliberti, 2025-09-25:
Hi @chiahao3,

See the cell output following the initialisation step: init = Initializer(init_params).init_all().

2025-09-25 14:24:28,490 - ### Initializing measurements ###
2025-09-25 14:24:28,492 - Loading measurements from source = 'file'
2025-09-25 14:24:28,493 - Detected measurement file type = '.h5'
2025-09-25 14:46:24,351 - Success! Loaded hdfs file with key = '/data' from path = 'data/ApoF/20240419_00848_Apo_CLA1-2_movie.h5'
2025-09-25 14:46:24,354 - Imported .hdf5 data shape = (262144, 512, 512)
2025-09-25 14:46:24,355 - Imported .hdf5 data type = uint16

MemoryError                Traceback (most recent call last)
Cell In(5], line 1
--> 1 init = Initializer (init params).init all()

File ~/anaconda3/envs/ptyrad/lib/python3.12/site-packages/ptyrad/initialization.py:594, in Initializer.init_all(self)
590 def init all(self):
591            # Run this method to initialize all
593           self.init cache()
--> 594    self.init measurements ()
595           self.init calibration()
596           self.set_variables_dict()

File ~/anaconda3/envs/ptyrad/lib/python3.12/site-packages/ptyrad/initialization.py:101, in Initializer.init measurements(self)
98 def init measurements(self):
99            vprint("### Initializing measurements ###", verbose=self.verbose) 
--> 101    meas = self._ load meas()
102          meas = self._process meas (meas)
104          meas avg = meas.mean(0) # This is equivalent to PACBED in electron microscopy. Note that if pad/resample are set to "on_the_fly", this would be different from the final one used for reconstruction.

File ~/anaconda3/envs/ptyrad/lib/python3.12/site-packages/ptyrad/initialization.py:703, in Initializer._load meas (self)
700 else:
701         raise ValueError (f"Unsupported measurement source '(meas_source)'. Use 'custom' or 'file'.*)
- -> 703 meas = meas. astype (' float32')
704         vprint ("Casting measurements dtype to float32 (single precision) for computational efficiency.")
705         vprint (f"Imported meausrements shape / dtype = (meas.shape), dtype = (meas.dtype)", verbose=self.verbose)

MemoryError: Unable to allocate 256. GiB for an array with shape (262144, 512, 512) and data type float32

Just to clarify: the input .h5 file is compressed to ~14.4 GB from .mrc (~137 GB) - with lossless compression.

Of course binning or cropping in k-space is an option but if at all possible I would like to avoid it - thus I haven't uploaded the summary_forward_pass_iter00XX.png for the successful run (btw that is 256x256x256x256 vs 512x512x512x512). 

Let me know if you need more info!

Thanks for the help!

Emanuela

chiahao3, 2025-09-25:
Hi @emanuelaliberti ,

Thanks for the log! Overall we can enable reconstructing large datasets with roughly 2 new features with source code changes (likely in future releases), or a highly customized script to manually implement those. Personally I'm more interested in integrating a long term solution to official PtyRAD, but if you'd like to explore the customized script option I'm happy to assist as well. 

---

Here're my notes for relevant points:

## File compression

Although the file compression can reduce the file when statically stored on disk, for any computation it'll need to be decompressed and loaded into RAM. So (262144,512,512) of uint16 would still take 512^4 * 2 bytes / 1024^3 = 128 GB in RAM. 

The fact that the dataset can be losslessly compressed into 14.4 GB with hdf5 means there're likely a lot of redundancy in the data. Assuming you're using the standard gzip compression for hdf5, which compress the data by replacing repeated sequence (i.e, 'aaaa' would be replaced as '4a'), I guess your dataset may contain a lot of contiguous 0 that would allow us to perform some k-space cropping without losing information.

## Data type casting in initialization

Ideally we don't want to sidestep the Initializer so you can still exploit the full flexibility with PtyRAD initialization. Therefore, I think the quickest solution for initialization is to perform all preprocessing in half precision (float16), which would still take 128 GB in RAM but seems like your machine has at least 145 GB available RAM for the successful run, this should be doable. I can add the option to toggle data types during initialization in future release, but as I mentioned previously that reconstruction with AD would usually require float32 for numerical stability so at some point we'll need to cast it to float32.

## Reconstruction and data loading into GPU VRAM

For reconstruction, currently PtyRAD would fully load the dataset into GPU VRAM for maximal speed so technically you can only load a dataset around 20GB into your V100 with 32GB VRAM. The multiGPU setup is also built for maximal speed so it'll actually duplicate the entire model and dataset on each GPU so we can run computation in parallel, see [DDP](https://docs.pytorch.org/tutorials/beginner/ddp_series_theory.html) if you're interested in the details. PyTorch has a built-in solution for dataset larger than GPU VRAM called `DataLoader` and I haven't fully implement it into PtyRAD yet, but presumably this would allow us to only load the necessary part into GPU VRAM during each iteration, and is necessary for large dataset although it'll be slower than loading everything into GPU VRAM.

## Cropping / Binning / Exporting

I understand you'll want to avoid cropping or binning in real / k-space, although I think binning in k-space should be quite doable and a binning factor of 2 or 3 might even increase the reconstruction quality due to higher SNR, as long as we don't hit aliasing artifact in the Ronchigram. Based on your V100 I think 256^4 would be the maximal dataset you can reconstruct without any code change.

For `meas_resample`, you'll set the following in your .yml. More descriptions are in the comments.
```yaml
'meas_resample'          : {'mode': 'precompute', 'scale_factors': [0.5,0.5]},
```

Just note that the float32 casting happens right after loading so you'll still hit the OOM if you're trying the 150 GB dataset. PtyRAD provides a series of preprocessing steps (i.e., cropping, resampling, flipping, etc.) so that users only need to keep one raw dataset to test different conditions, and they can optionally export the initialized dataset as hdf5 using `meas_export`.

chiahao3, 2025-11-25:
Hi @emanuelaliberti ,

No worries, I was busy working on job application as well, still slowly cruising.

Glad to hear that you've tested it on the 5GB dataset and it worked fine! Regarding the 2 test cases:

## Test 1
Took me a while to pinpoint the issue because the `problems.txt` only shows the final error traceback. I should've asked for the full log or a minimal test dataset but luckily I think I've reached a reasonable explanation. This error probably stem from the extremely limited dynamic range of float16 dtype, which is only ~ 65535 and any number larger than this will be represented as `inf` in float16. 

Although I don't have the full log for Test 1, we can see that the mean value of the full dataset (262144,512,512) is 0.4466 from the Test 2 log (line 193). This suggests the **total intensity of the averaged diffraction pattern** is 512x512x0.4466=117073, which is larger than the maximum legal value of float16 (~65535). This total intensity of the averaged diffraction pattern (dataset.mean(0)) is used to normalize the probe intensity, so it's causing illegal values in the probe during initialization. I suspect if you check the full log during initialization, you might see there're NaN values in the probe. I've pushed a simple fix to the `dataset` branch and it should solve Test 1, which basically just calculate such normalization constant in float32.

**Solution:** Download the latest `dataset` branch (Commit b2db6ae) and it should run smoothly for `meas_dtype` = 'float16'. If not, send me the full log, a test dataset, and we can arrange a Zoom chat sometime to figure this out.

## Test 2
It's great that you managed to find a machine with ~440 GB RAM, but in this specific case (casting from uint16 to float32 inside PtyRAD) it's still not enough, because it'd try to first allocate 137x2 = 274 GB memory for the float32 array, and then copy over the data from the original 137GB dataset in uint16 already loaded in RAM. Therefore, this operation would cost at least 137x3 = 411 GB, hence causing memory issues. 

**Solution:** However, you can surely write a script to convert the 137 GB dataset into float32 in a memory map way and save it on disk, which will generate a 274 GB dataset. You can then load this dataset directly into PtyRAD and keep it at float32, this way it'd run smoothly given you have ~400GB RAM. 

## Thoughts on long term solutions
Currently PtyRAD (the `dataset` branch) streams the data from memory into GPU memory in batches so we're not limited by GPU memory anymore, but apparently RAM has become the next obvious bottleneck due to those preprocessing operations. It's possible to avoid applying those operations to the whole dataset, and only apply them on-the-fly while loading batches, but that'd require quite some refactoring of the current codebase. I'm generally interested in such changes to enable hundred GB or even TB dataset reconstructions, but that might not happen until next Spring. It's more likely to come up with a temporary script for such needs though.

chiahao3, 2025-11-26:
Hi @piperfw, @emanuelaliberti ,

No problem, I'm sure we all want to resolve this as soon as possible! Thanks you two for getting back on this so quickly too!

I managed to reproduce the error (Test 1) locally and identified the issue. It's the final `meas.astype()` trying to allocate a new array even though it's already in float32. The fix is rather simple with `copy=False` and I pushed it to `dataset` already (Commit efc0339). You should be able to just download the latest `dataset` branch, and use `'meas_dtype' : 'float32'` with `'preload_data' : false` on your float32 dataset. Let me know if it works! There're still a couple small adjustment I want to make but with these valuable bug reports I'm confident we can merge this to main branch soon.

## Successful reconstruction with the fix
Synthesized (384,128,512,512) float32 dataset (48GB), tested it on Win10 workstation with 64GB RAM and 16GB VRAM (Quadro P5000).
You can see that the memory usage is tight but eventually worked out fine. This dataset is larger than GPU VRAM so we can see that the asynchronous loading is working correctly as well.

*10sec GIF that spans for 10min, you might need to press Play on the top right corner*
[image]

[20251126_142140_ptyrad_log.txt](https://github.com/user-attachments/files/23780009/20251126_142140_ptyrad_log.txt)
[tBL_WSe2_large.yml](https://github.com/user-attachments/files/23780007/tBL_WSe2_large.yml)
[save_hdf5.ipynb](https://github.com/user-attachments/files/23780164/save_hdf5.ipynb)

## Reproduced error (Test 1) without the fix
[image]

I realized I wasn't being super clear about what I meant by "log". The log files are stored with the reconstruction output by design, and you're 100% correct that if PtyRAD errors out during initialization, it won't produce the typical txt log. 

For future reference (and for other users following this thread):

In order to debug these early failure cases, you can: 
1. Use a jupyter notebook (like the `run_ptyrad_detailed_walkthrough.ipynb` in demo/scripts) to execute PtyRAD and save all the cell output and then send it. You may also manually copy/paste the cell outputs and error tracebacks into a text file. It's always better to have the full cell output in addition to the error traceback, because some error might be generated early but not causing issue until later (like the NaN in probe initialization we had in float16).
2. If you're running PtyRAD with Slurm, usually there's a standard output log generated by Slurm and it can be specified in the .sbatch script like `#SBATCH --output=log_20251126_ptyrad.txt  # Standard output and error log`. See https://github.com/chiahao3/ptyrad/blob/main/demo/scripts/slurm_run_ptyrad.sub for example.
3. If you're running PtyRAD from Linux bash, you can run: 
```
## Execute the ptyrad CLI command `ptyrad run`
ptyrad run --params_path <PARAMS_PATH> --gpuid 0 > ptyrad_log.txt 2>&1
```
Replace `<PARAMS_PATH>` with the actual path to yml params file. The `> ptyrad_log.txt 2>&1` would redirect the python output to a text file and save it under your working directory.

emanuelaliberti, 2025-12-04:
Hi @chiahao3 

Thanks for your help in making progress with these large datasets. I would share the reconstruction results with you if the recovered object looked as expected.

As you can see from the log file I shared, the solver reached convergence after approximately 1 day, 8 hr 32 min for a realistic number of 200 iterations.

For smaller datasets, we would now proceed with hyperparameter tuning adjusting `defocus` and `rotation` first, as these typically have the strongest effect on convergence.

However, with datasets of this size, this process becomes extremely time consuming and, in the long run, not scalable.

Since you mentioned the parallax approach, we have been considering a hybrid workflow where we first run parallax to fit defocus and rotation, and then run PtyRAD.

To make parallax fast, we have been exploring a pixel-wise streaming data-loading approach (similar in concept to a PyTorch DataLoader) rather than loading all frames into memory at once.

I therefore have a few questions:

1. Are there any plans to include parallax (tbBF) into the PtyRAD workflow? I noticed that tbBF? appears under the Initialization section in your WISHLIST.md, so I am wondering whether integration is planned.

2. Are there any plans to make defocus and scan rotation AD-optimisable? In an older issue (#19) you mentioned that the AD probe update can compensate some defocus error, but only marginally. This is insufficient in our case because our defocus values are on the micrometre scale, and we work at biological dose levels. Since defocus describes the probe propagation (forward operator) and scan rotation the geometry, I am curious why these 2 parameters are not currently included in the AD optimisation loop.

3. Have you explored whether multi-GPU setups could help with large-scale hyperparameter searches? So far, I have not found a “smart” strategy that scales well for other parameters (learning rates, etc..).

4. Related to another potential issue: I have not yet found a reliable way to male PtyRAD more serialisable. Using Optuna for hyperparameter optimisation still requires significant manual visual inspection, and simply restarting PtyRAD with the best candidates has not yet produced trustworthy results. If there are any recommended practices for producing reliable or serialisable optimisation states, I would appreciate these.

As I mentioned before, we are very keen to collaborate and help test new releases and developments.

Looking forward to your thoughts!

[20251128_154850_ptyrad_log.txt](https://github.com/user-attachments/files/23931912/20251128_154850_ptyrad_log.txt)